zynthian/zynthian-sys #103: Reduce quantity of (apparently duplicate) synth engines

riban-bw, 2019-05-23:
The list of synth engines presented when adding new synth layer is very long which may prove difficult to navigate. There are many entries which seem to relate to Fluidsynth which provide just one instrument but not necessarily a useful or the expected one.

It would be useful to minimise the list so that it is more easily navigated. It would further be useful to explain the rationale for what is (and is not) in the list. Maybe that could be on the wiki. If there are variations of the same engine then it may be more appropriate to add these to a sub-menu, e.g. the patch selection of a layer.

It can be confusing if similar options are available via different routes and it can detract from the flexibility and excellence of a product if there are too many choices, especially if those choices provide little benefit to most users.

jofemodo, 2019-05-23:
I've transferred to zynthian-sys because the default config file for Jalv engines is managed here.

jofemodo, 2019-05-30:
I think that it's better to give to "first time" users a good overview of synth engines options, although it's a little bit redundant. Reducing the list is very easy from webconf ...

riban-bw, 2019-05-30:
I understand your idea to expose the wide range available but I don't understand what these fluidsynth presets are and it hadn't occurred to me that I can remove them. Maybe we need a bit more about this in the user guide

riban-bw, 2019-06-08:
I have just had another look at this and I still don't understand what all the Fluid plugins are supposed to be representing. I understand that we should provide a wide range of examples for users to experiment with and let them know that they can enable / disable plugins from the webconf but, if like me, other users are simply confused by the sheer quantity of plugins and do not understand what they represent then we are doing ourselves a disservice. 

I observe that many jalv plugin URLs do not resolve in a web browser which makes it difficult to trace documentation for these. I assume somewhere there is a map of URI to actual plugin location.

Is someone able to explain what each of these Fluid Plugins is supposed to represent? I cannot see their benefit.

jofemodo, 2019-11-15:
Ahh! You can see the new list in: 0597350

jofemodo, 2019-11-15:
OK! For updating this on working systems you have to delete the current list:

rm -f /zynthian/config/jalv/*

and after that, update the software.

closed 2019-11-15 by jofemodo