zynthian/zynthian-sys #106: NTP and "Time has been changed" causes Jack Shutdown

smiths73v3, 2019-06-08:
Related to #85, I think this is the root cause of a number of problems I was seeing on the "first boot of the day"

A large time jump between initial system time, and when the network interface (wifi in my case) becomes available causes a Jack timeout. At this point, Jack is down, and does not come back online. Either a restart of the UI, or a reboot will restore Jack and avoid a large time jump due to the recent system time update.

Sorry, no suggestions at the moment for how to solve this one.....

```
Jun 08 04:56:58 zynthian dhcpcd[692]: wlan0: leased 192.168.1.153 for 21600 seconds                                             
Jun 08 04:56:58 zynthian avahi-daemon[401]: Joining mDNS multicast group on interface wlan0.IPv4 with address 192.168.1.153.    
Jun 08 04:56:58 zynthian avahi-daemon[401]: New relevant interface wlan0.IPv4 for mDNS.                                         
Jun 08 04:56:58 zynthian avahi-daemon[401]: Registering new address record for 192.168.1.153 on wlan0.IPv4.                     
Jun 08 04:56:58 zynthian dhcpcd[692]: wlan0: adding route to 192.168.1.0/24                                                     
Jun 08 04:56:58 zynthian dhcpcd[692]: wlan0: adding default route via 192.168.1.1                                               
Jun 08 04:57:06 zynthian startx[459]: ERROR:root:Jack Audio XRUN!                                                               
Jun 08 04:57:06 zynthian startx[459]: ERROR:root:Jack Audio XRUN!                                                               
======= Large jump in time caused by NTP sync with timeserver after WIFI online =====
Jun 08 15:25:57 zynthian systemd[1]: Time has been changed                                                                      
Jun 08 15:25:57 zynthian a2jmidid[533]: JackPosixSemaphore::TimedWait err = Connection timed out                                
Jun 08 15:25:57 zynthian a2jmidid[533]: SuspendRefNum error                                                                     
Jun 08 15:25:57 zynthian a2jmidid[533]: JackClient::Execute error name = a2j                                                    
Jun 08 15:25:57 zynthian ttymidi[537]: JackPosixSemaphore::TimedWait err = Connection timed out                                 
Jun 08 15:25:57 zynthian ttymidi[537]: SuspendRefNum error                                                                      
Jun 08 15:25:57 zynthian ttymidi[537]: JackClient::Execute error name = ttymidi                                                 
Jun 08 15:25:57 zynthian systemd-timesyncd[307]: Synchronized to time server 163.237.218.19:123 (2.debian.pool.ntp.org).        
Jun 08 15:25:57 zynthian systemd[1]: apt-daily.timer: Adding 7h 33min 15.922496s random time.                                   
Jun 08 15:25:57 zynthian systemd[1]: apt-daily-upgrade.timer: Adding 8min 55.270555s random time.                               
Jun 08 15:25:57 zynthian startx[459]: JackPosixSemaphore::TimedWait err = Connection timed out                                  
Jun 08 15:25:57 zynthian startx[459]: SuspendRefNum error                                                                       
Jun 08 15:25:57 zynthian startx[459]: JackClient::Execute error name = ZynMidiRouter                                            
Jun 08 15:25:57 zynthian startx[459]: JackPosixSemaphore::TimedWait err = Connection timed out                                  
Jun 08 15:25:57 zynthian startx[459]: SuspendRefNum error                                                                       
Jun 08 15:25:57 zynthian startx[459]: JackClient::Execute error name = jackpeak                                                 
Jun 08 15:25:57 zynthian systemd[1]: Reached target Network is Online.                                                          
Jun 08 15:25:57 zynthian a2jmidid[533]: WARNING: JACK server shutdown notification received.                                    
Jun 08 15:25:57 zynthian systemd[1]: Starting Daily apt download activities...                                                  
Jun 08 15:25:57 zynthian a2jmidid[533]: Bridge stopping...                                                                                                                                                                                                   
```

jofemodo, 2019-06-10:
This issue is already addressed in zynthian-sys#93 and should be solved in latest builds (post-RC2).

closed 2019-06-10 by jofemodo

smiths73v3, 2019-06-10:
Excellent! Good to hear this is solved.