zynthian/zynthian-sys #112: Build hangs on compliation of mod-distortion

guysoft, 2019-07-17:
```
make[1]: Leaving directory '/home/pi/zynthian-sw/plugins/mod-distortion/ds1'
/usr/bin/make -C bigmuff install
make[1]: Entering directory '/home/pi/zynthian-sw/plugins/mod-distortion/bigmuff'
mkdir -p /zynthian/zynthian-plugins/lv2/mod-bigmuff.lv2
cp mod-bigmuff.so /zynthian/zynthian-plugins/lv2/mod-bigmuff.lv2
cp -r ttl/* /zynthian/zynthian-plugins/lv2/mod-bigmuff.lv2
make[1]: Leaving directory '/home/pi/zynthian-sw/plugins/mod-distortion/bigmuff'
/usr/bin/make -C ds1 clean
make[1]: Entering directory '/home/pi/zynthian-sw/plugins/mod-distortion/ds1'
rm -f src/*.o mod-ds1.so
make[1]: Leaving directory '/home/pi/zynthian-sw/plugins/mod-distortion/ds1'
/usr/bin/make -C bigmuff clean
make[1]: Entering directory '/home/pi/zynthian-sw/plugins/mod-distortion/bigmuff'
rm -f src/*.o mod-bigmuff.so
make[1]: Leaving directory '/home/pi/zynthian-sw/plugins/mod-distortion/bigmuff'
Cloning into 'mod-pitchshifter'...
Generating harmonizer.wisdom file, this might take a while...
fftwf-wisdom -v -n -x -o Shared_files/harmonizer.wisdom \
rof128 rob128 rof192 rob192 rof256 rob256 rof384 rob384 rof512 rob512 rof640 rob640 rof768 rob768 \
rof1024 rob1024 rof1280 rob1280 rof1536 rob1536 rof2048 rob2048 rof2176 rob2176 rof2304 rob2304 \
rof2432 rob2432 rof2560 rob2560 rof3072 rob3072 rof4096 rob4096
Build timed out (after 360 minutes). Marking the build as failed.
Build was aborted
```

Generating harmonizer.wisdom file, this might take a while seems to hang for 6 hours and dies.

jofemodo, 2019-07-17:
Yes, i noted that. It was working nicely on stretch, but something is broken now ...
I will try to fix it ASAP ...

jofemodo, 2019-07-18:
I had a fix that could work, but when trying to cancel the current build for launching a new one,  the things didn't work and build processes remained detached from Jenkins.

I rebooted the machine, but then. again the:

`02:48:54 losetup: 2019-07-10-raspbian-buster-lite.img: failed to set up loop device: No such file or directory
`
Ups! ;-)

guysoft, 2019-07-18:
ran
```
root@zynthianOS:/home/guysoft/ZynthianOS/src# docker exec -it zynthianos-build bash
root@897a1c5d43b4:/# losetup -f    
/dev/loop0
```

Then started again just now and it seems to work.

Seems strange, 
That error should show up either when it fails to create the /dev/loop device (which it didn't because in my shell output above you can see it returns ``/dev/loop0``). Or if the image does not exist, which it did).

Looks like its worth adding a test to make sure the file is there before running the losetup command to make sure the image is there, and rule out the latter option. Might let us zoom on to the issue.

jofemodo, 2019-07-19:
Failing again. I did:

root@zynthianOS:/home/guysoft/ZynthianOS/src# docker exec -it zynthianos-build bash
root@897a1c5d43b4:/# losetup -f    
/dev/loop0

Then started again just and didn't work.

;-(

jofemodo, 2019-07-31:
HI @guysoft !

This is very frustrating! The build system is broken from 2 weeks ago. The error is the same error we had in the past all the time:

```
losetup -f --show -o 276824064 2019-07-10-raspbian-buster-lite.img
losetup: 2019-07-10-raspbian-buster-lite.img: failed to set up loop device: No such file or directory

```

And i don't understand where the problem is. I followed your tips, but nothing seems to work. I'm frustrated and embarrased, but i really need your help for solving this problem "forever". AFAIK, It's a silly error and shouldn't be breaking things all the time ...

Thanks a lot for your time!

guysoft, 2019-07-31:
It might be related to moving to buster and the loopback stuff set
different there. Ill try and find to investigate. Realistically after the
4th of August.

commit 20a250f6 (2019-07-18): Second try to fix #112: The failing recipe is mod-shifter, not mod-distortion ;-)
closed 2019-09-03 by jofemodo