zynthian/zynthian-sys #115: Wifi problem on raspberry4 buster

lguyome45, 2019-08-17:
Hi,
with the nightly build, wifi is not working, wlan0 does not exist.
I take a look into the file setup_system_rbpi_raspbian_buster.sh and see the installation of a non-free wifi firmware.
In the build.log of the creation on ZynthianOs when installing the firmware, system tell that we downgrad the firmware.
I just make an apt install firmware-brcm80211 and after a reboot wifi is working

For Raspberry 4 we need to remove 

> # Non-free WIFI firmware for RBPi3
> wget https://archive.raspberrypi.org/debian/pool/main/f/firmware-nonfree/firmware-brcm80211_20161130-3+rpt3_all.deb
> dpkg -i firmware-brcm80211_20161130-3+rpt3_all.deb
> rm -f firmware-brcm80211_20161130-3+rpt3_all.deb

jofemodo, 2019-09-03:
It's merged ...

lguyome45, 2019-09-05:
I test the new image with a Rbpi3B+ and wifi works.
I just have a problem on both raspberry where sometimes the wifi is not enabled, need to go to the admin panel to activate

jofemodo, 2019-09-05:
OK! This problem is a different one. It's present in Aruk also. It should be addressed in a different issue.
Let's close this one ;-)

closed 2019-09-05 by jofemodo