zynthian/zynthian-sys #121: WIFI "sometimes" doesn't get enabled on boot.

jofemodo, 2019-09-05:
The problem is present in Aruk and also in the new Buster images.
Probably, it was  introduced with the new WIFI setup script.

mheidt, 2019-11-10:
And could we get the HOTSPOT enabled in the initial image?

jofemodo, 2019-11-10:
It's very insecure and everybody should remember to disable it or change the password.
Official zynthian images won't have the hotspot enabled by default.

jofemodo, 2019-11-10:
What we could do is to enable the hotspot under some conditions:

+ No other network is enabled
+ Zynthian UI is on ERROR status

Or something like that ... let's think a little bit about it ;-)

mheidt, 2019-11-10:
Could we have a trigger file in root? You can't access webconf on non V3 kits if you don't have a ethernet router Sunday,

jofemodo, 2019-11-12:
We already have it ... more or less ;-)
Simply set:

- ZYNTHIAN_WIFI_MODE=hotspot 

in the config file "/zynthian/config/zynthian_envars.sh". You could create a script to do it quickly after burning an SD card.

Regards,.

mheidt, 2019-11-12:
But it has to be something a windows user can do quickly when mounting the freshly burned image.
He or she can't run a script that easily.
I would have a file in root, that says DELETE_FOR_HOTSPOT.MD
with a more detailed description inside.
At first boot, the existence is checked and if it is gone, we can call that script you suggested.

jofemodo, 2019-11-12:
People building custom zynthians normally doesn't have problems with adding a line to a text file.
It's not much more complex than creating/copying a file to a folder.

Anyway, you could connect your zynthian to your PC using an ethernet point-to-point connection, access webconf and turn-on the hotspot if you want. 

In both cases (hotspot or ethernet point2point), you end using the computer for accessing the webconf. I really don't see the need for the hotspot on the first boot. A simple ethernet cable works perfectly and don't have security problems.

Regards,

mheidt, 2019-11-12:
I am not talking about me and the other "maker". I had those v1-v2+ kit users in mind.
We met people in student housing that have  wifi access only as well.
I thought this feature would be an easy method to give support as well.

But nvm, I have to get the webconf working on an gorgona build:
  File "/zynthian/zynthian-webconf/lib/dashboard_handler.py", line 216, in get_gpio_expander
    if out[3].startswith("20: 20"):
IndexError: list index out of range

mheidt, 2019-11-12:
fixed...

jofemodo, 2019-12-29:
It should be solved now. Please update & test!

closed 2019-12-29 by jofemodo