zynthian/zynthian-sys #122: Buster / Audioinjector Ultra / Raspi4

mheidt, 2019-10-08:
I had a running system. An early buster image.
Once I updated, it didn't start anymore.
I took the image 10-08 and started new.
I could configure it via webconf 
But it still fails

Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
libjackpeak failed to start jack client: 17
convert-im6.q16: unable to open image `/zynthian/config/img/zynthian_logo_error_ip.png': No such file or directory @ error/blob.c/OpenBlob/2874.
convert-im6.q16: WriteBlob Failed `/zynthian/config/img/zynthian_logo_error_ip.png' @ error/png.c/MagickPNGErrorHandler/1641.
/zynthian/config/img/zynthian_logo_error_ip.png: image not found
didn't find evidence of prior run.
  Building XImage...done
^Cxinit: connection to X server lost

waiting for X server to shut down (II) Server terminated successfully (0). Closing log file.

xinit: unexpected signal 2

What can I do to get more information?

jofemodo, 2019-10-08:
What audioinjector model are you using? It seems that Jack can't start. Perhaps the jackd configuration doesn't world with your card ...

mheidt, 2019-10-08:
Audioinjector Ultra.
root@zynthian:~# alsamixer
cannot open mixer: No such file or directory
root@zynthian:~# amixer
amixer: Mixer attach default error: No such file or directory

Does this give a hint?

jofemodo, 2019-10-08:
Is the kernel driver loaded? What is the output of lsmod?

mheidt, 2019-10-08:
root@zynthian:~# lsmod
Module                  Size  Used by
xt_state               16384  0
xt_conntrack           16384  0
nf_conntrack          135168  2 xt_state,xt_conntrack
nf_defrag_ipv6         20480  1 nf_conntrack
nf_defrag_ipv4         16384  1 nf_conntrack
nft_counter            16384  1
nft_chain_route_ipv4    16384  1
xt_DSCP                16384  1
xt_tcpudp              16384  1
nft_compat             20480  2
nf_tables             122880  7 nft_compat,nft_chain_route_ipv4,nft_counter
nfnetlink              16384  2 nft_compat,nf_tables
8021q                  32768  0
garp                   16384  1 8021q
stp                    16384  1 garp
llc                    16384  2 garp,stp
evdev                  24576  0
joydev                 20480  0
snd_usb_audio         192512  1
snd_hwdep              16384  1 snd_usb_audio
snd_usbmidi_lib        32768  1 snd_usb_audio
fb_ili9486             16384  0
snd_rawmidi            32768  1 snd_usbmidi_lib
fbtft                  45056  1 fb_ili9486
hid_multitouch         24576  0
brcmfmac              311296  0
snd_seq_device         16384  1 snd_rawmidi
syscopyarea            16384  1 fbtft
sysfillrect            16384  1 fbtft
sysimgblt              16384  1 fbtft
fb_sys_fops            16384  1 fbtft
brcmutil               16384  1 brcmfmac
ads7846                24576  0
snd_soc_pcm512x_i2c    16384  0
snd_soc_pcm512x        32768  1 snd_soc_pcm512x_i2c
sha256_generic         20480  0
bcm2835_codec          36864  0
bcm2835_v4l2           45056  0
v4l2_mem2mem           24576  1 bcm2835_codec
cfg80211              614400  1 brcmfmac
snd_soc_dmic           16384  0
bcm2835_mmal_vchiq     32768  2 bcm2835_codec,bcm2835_v4l2
rfkill                 28672  2 cfg80211
snd_soc_hifiberry_dacplusadc    16384  0
v4l2_common            16384  1 bcm2835_v4l2
snd_soc_bcm2835_i2s    20480  0
videobuf2_dma_contig    20480  1 bcm2835_codec
videobuf2_vmalloc      16384  1 bcm2835_v4l2
videobuf2_memops       16384  2 videobuf2_dma_contig,videobuf2_vmalloc
snd_soc_core          192512  4 snd_soc_pcm512x,snd_soc_hifiberry_dacplusadc,snd_soc_bcm2835_i2s,snd_soc_dmic
videobuf2_v4l2         24576  3 bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem
snd_compress           20480  1 snd_soc_core
videobuf2_common       45056  4 bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
snd_pcm_dmaengine      16384  1 snd_soc_core
snd_pcm               102400  6 snd_soc_pcm512x,snd_usb_audio,snd_soc_hifiberry_dacplusadc,snd_pcm_dmaengine,snd_soc_bcm2835_i2s,snd_soc_core
videodev              200704  6 bcm2835_codec,v4l2_common,videobuf2_common,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
snd_timer              32768  1 snd_pcm
media                  36864  2 videodev,v4l2_mem2mem
snd                    73728  12 snd_compress,snd_soc_pcm512x,snd_hwdep,snd_usb_audio,snd_timer,snd_rawmidi,snd_usbmidi_lib,snd_seq_device,snd_soc_core,snd_pcm
raspberrypi_hwmon      16384  0
vc_sm_cma              36864  1 bcm2835_mmal_vchiq
i2c_bcm2835            16384  0
hwmon                  16384  2 raspberrypi_hwmon,ads7846
spi_bcm2835            20480  0
rpivid_mem             16384  0
clk_hifiberry_dacpro    16384  0
uio_pdrv_genirq        16384  0
uio                    20480  1 uio_pdrv_genirq
fixed                  16384  0
i2c_dev                20480  0
ip_tables              24576  0
x_tables               32768  6 xt_state,ip_tables,nft_compat,xt_tcpudp,xt_DSCP,xt_conntrack
ipv6                  450560  26

jofemodo, 2019-10-08:
Hifiberry DAC+ADC is loaded instead of audioinjector. Could you send a screenshot of webconf's Dashboard?

mheidt, 2019-10-08:
[image]
I changed kit to custom and all other setting without a reboot..only after I changed everything...

mheidt, 2019-10-08:
I changed kit back to v3, custom, reboot, audio card to ultra, reboot..

Still:
snd_soc_core          192512  4 snd_soc_pcm512x,snd_soc_hifiberry_dacplusadc,snd_soc_bcm2835_i2s,snd_soc_dmic

jofemodo, 2019-10-08:
Could you try to run update_zynthian_sys.sh from command line?

mheidt, 2019-10-08:
Updating System configuration ...
SOUNDCARD CONFIG => dtoverlay=audioinjector-ultra
DISPLAY CONFIG => dtoverlay=piscreen2r-notouch,rotate=270\ndtoverlay=ads7846,speed=2000000,cs=1,penirq=17,penirq_pull=2,swapxy=1,xohms=100,pmax=255
ln: failed to create symbolic link '/zynthian/zynthian-sw/pianoteq6/pianoteq': No such file or directory
Aeolus Waves already exist!
Custom Config /zynthian/zynthian-sys/custom/display/ZynScreen 3.5 (v1) ...

(I haven't changed the screen yet)

Could it be, that the reboot from webconf is not working anymore?

When I rebooted from ssh, the alsamixer is showing!

mheidt, 2019-10-08:
and now, the webconf reboot works again

mheidt, 2019-10-08:
Did you find anything, why I had to run it manually? Should I leave this issue open?