zynthian/zynthian-sys #124: Missing liblilv-dev

guysoft, 2019-10-28:
See:

https://jenkins.zynthian.org/job/ZynthianOS/677/parsed_console/

jofemodo, 2019-10-30:
It should be solved by 74e30a1

mheidt, 2019-11-10:
No, it is not solved.
I will test again with a network connection during first boot.
But the goal is, that the first boot doesn't need a network connection and that the HOTSPOT is turned on.

jofemodo, 2019-11-10:
I don't see any problem in the last green builds. Lilv is compiled and installed correctly.
Please, be more specific about the SD-image and the error itself.

mheidt, 2019-11-10:
I burned 11-08 and after initial boot of a v2 kit there is no webconf running.
I tried both: either with or without internet connection.
When starting manually:
root@zynthian:/zynthian/zynthian-webconf# ./zynthian_webconf.sh 
INFO:root:ZYNTHIAN-UI CONFIG ...
INFO:root:Wiring Layout MCP23017_ZynScreen
DEBUG:root:ZYNCODER A: [102, 105, 110, 113]
DEBUG:root:ZYNCODER B: [101, 104, 109, 112]
DEBUG:root:SWITCHES layout: [100, 103, 108, 111, 106, 107, 114, 115]
DEBUG:root:DISABLED_MIDI_IN = ['']
DEBUG:root:ENABLED_MIDI_OUT = ['ttymidi:MIDI_out']
DEBUG:root:ENABLED_MIDI_FB = ['']
DEBUG:root:MMC Bank Change CCNum: 32
DEBUG:root:MMC Bank Change UP: None
DEBUG:root:MMC Bank Change DOWN: None
DEBUG:root:MMC Program Change UP: None
DEBUG:root:MMC Program Change DOWN: None
WARNING:root:No Display!
Traceback (most recent call last):
  File "./zynthian_webconf.py", line 55, in <module>
    from lib.jalv_lv2_handler import JalvLv2Handler
  File "/home/pi/zynthian-webconf/lib/jalv_lv2_handler.py", line 29, in <module>
    import lilv
ModuleNotFoundError: No module named 'lilv'

Besides that. What do I have to do to get it running? it's not pip3 install lilv

jofemodo, 2019-11-10:
It seems like the lilv install script not copying the lilv python modules.
I will solve it tomorrow ...

wyleu, 2019-11-10:
It complains of no tornado.web which pip install tornado cures but then
requires other python modules till it asks for jack and then fails on the
install....

mheidt, 2019-11-16:
issue still exists in 11-14

jofemodo, 2019-11-18:
Yes. You are right. I just fixed the problem (i hope!!).
For fixing a previous "green" Buster image, you simply run:

```
update_zynthian.sh
$ZYNTHIAN_RECIPE_DIR/install_lv2_lilv.sh
```

It should work ;-)

commit 4414d456 (2019-11-18): Fix #124: Fix lilv installation to copying python library to the right place.
closed 2019-11-18 by jofemodo