zynthian/zynthian-sys #126: Update stalls installaing amsynth

riban-bw, 2019-11-29:
**Describe the bug**
Performing an update from the Zynthian UI stalls (for at least 9 hours!!!) building amsynth.

**To Reproduce**
Steps to reproduce the behavior:
1. From Admin menu select "Update Software

**Expected behaviour**
Each software element is updated as required then UI returns to normal operational mode.

**Actual behaviour**
Some software elements update. amsynth starts to install (compile) and processes through many compilation stages but gets stuck compiling the main app:
```
CXXLD  amsynth_dssi.la
CXXLD  amsynth_lv2l.a
CXXLD  amsynth_lv2_gtk.la
CXXLD  amsynth
```

**Setup:**
SD-image
Buster nightly 2019-10-20 updated 2019-11-27

HARDWARE
Raspberry Pi 3 Model B Plus Rev 1.3
Soundcard: HifiBerry DAC+ light
Display: Sainsmart 1.8
Wiring: I2C_HWC
GPIO Expander: No detected

SOFTWARE
zyncoder: master (8b7bfbd)
zynthian-ui: master (c02b3ce)
zynthian-sys: master (c16eb29)
zynthian-data: master (d7648f7)
zynthian-webconf: master (813b47d)

**Additional context**
I left it overnight (approx. 9 hours) without any change in behaviour. I ssh in and observed compile processes still running in process list. I rebooted and repeated update with similar results, stalling building amsynth. Process list shows:

```
bash───install_amsynth───make───make───bash───make─┬─bash───bash───g++───cc1plus
                                                   └─bash───bash───ar
```

jofemodo, 2019-11-29:
When updating from UI, stderr buffer is full and process stalls forever. Solved by piping stderr to stdout.

riban-bw, 2019-11-29:
This fix is not applied during update, i.e. zynthian_gui_admin.py is not updated when I perform an update from the Zynthian.

I think that zynthian_gui_admin.py is updated after plugins so the fault blocks this fix from being applied.

jofemodo, 2019-11-29:
Ups! It's true ...

I've redirected make's stderr to /dev/null. I tested and it seems to solve the problem for this update.
As UI has been fixed, we shouldn't have this kind of problems with future updates.

Thanks!

commit 610165cf (2019-11-29): Fixes #126: Send make's stderr to /dev/null for avoiding update problems from UI.
UI has been fixed, so it shouldn't be needed never more after this update.
closed 2019-11-29 by jofemodo