zynthian/zynthian-sys #129: Midi In Hickup with NoteOff/Vel0 and new ttymidi

mheidt, 2019-12-02:
Setup:
V2 Kit
D50 over ttymidi
Buster image

I am playing trills over 59/60 and it works when I use a non-buster image.
In those old images I see the D50 sending NoteOn Vel0.

But in buster they became NOTE-OFFs.
Apparently because of the new ttymidi change mentioned earlier in the forum.
Unfortunately there are hickups.

CH#00 CONTROL_CHANGE 123 => 0
CH#00 NOTE_OFF 59, Vel: 64
CH#00 NOTE_ON 73, Vel: 59
CH#00 NOTE_OFF 123, Vel: 64
CH#00 NOTE_ON 60, Vel: 88
CH#00 NOTE_OFF 60, Vel: 64

I can understand the OFF 123....a bug with the value of the CONTROL_CHANGE...but I can't explain the 73.

I could confirm this behaviour on a second v1 MCP23008 zynthian as well.
Works well with pre buster and fails with buster.

mheidt, 2019-12-02:
https://github.com/moddevices/mod-ttymidi/issues/8

jofemodo, 2019-12-16:
@mheidt, i fixed the "building" problem on ttymidi:

https://github.com/jofemodo/mod-ttymidi

The PR is created, but we can advance in our side testing your patch, so please, proceed with it in this new fork. I changed the zynthian recipe for using this repository temporally, so we can test your changes integrated in the nightly builds ;-)

jofemodo, 2019-12-17:
It seems to be fixed in the official mod-ttymidi repo.
Can you confirm, @mheidt ?

mheidt, 2019-12-17:
yep

closed 2019-12-17 by mheidt