zynthian/zynthian-sys #133: FluidSynth Bank load broken

mheidt, 2019-12-23:
I took the image from 12-20, updated.
Using the FluidSynth doesn't work. 
ERROR:zynthian_engine: Can't exec engine command: load "/zynthian/zynthian-data/soundfonts/sf2/FluidR3_GM.sf2" => End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x6dcba8d0>
command: /usr/local/bin/fluidsynth
args: ['/usr/local/bin/fluidsynth', '-p', 'fluidsynth', '-a', 'jack', '-m', 'jack', '-g', '1', '-j', '-o', 'synth.midi-bank-select=mma', '-o', 'synth.cpu-cores=3', '-o', 'synth.polyphony=64']
buffer (last 100 chars): b''
before (last 100 chars): b'load "/zynthian/zynthian-data/soundfonts/sf2/FluidR3_GM.sf2"\r\n'
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 3346
child_fd: 30
closed: False
timeout: 20
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_re:
    0: re.compile(b'\n> ')
WARNING:zynthian_engine_fluidsynth: SoundFont '/zynthian/zynthian-data/soundfonts/sf2/FluidR3_GM.sf2' can't be loaded

jofemodo, 2019-12-23:
I think it could be related with #132. Could you check if if the '/zynthian/zynthian-data/soundfonts/sf2/FluidR3_GM.sf2' is a valid soft link and the destiny does exist?
Could you try to install/upload another SF2 ans see if it's loaded correctly?

Thanks

mheidt, 2019-12-23:
Actually I uploaded a sleigh bell soundfont when I had the issues. But the built-ins had the same issue.

mheidt, 2019-12-31:
It doesn't happen all the time but often. Once it  couldn't load a soundfont anymore, it won't load any other. Restarting the UI helps.
Do we have a new version of fluidsynth? There are no changes in the zyngine.
New python version?
I will retest with the latest image

jofemodo, 2020-01-15:
It seems that if you load a "bad" soundfont, fluidsynth engine fails and can't be recovered until engine is restarted. Could you confirm this behaviour?

mheidt, 2020-01-15:
I could make the build-in soundfont behave badly. Not all the time. But especially if you reboot with a fs-layer as last snapshot.

jofemodo, 2020-01-22:
I think the error is caused by a memory leak on fluidsynth code, probably introduced on v2.1.0.
I opened an issue:

    https://github.com/FluidSynth/fluidsynth/issues/613

Until it's solved, i've modified the install script for using v2.0.9. Please, re-build your fluidsynth and check if problem has gone ...

`$ZYNTHIAN_RECIPE_DIR/install_fluidsynth.sh `

Thanks

mheidt, 2020-01-22:
yes, v2.0.9 works for me

commit f4a811da (2020-01-22): Temporary fix #133: Use older v2.0.9 to avoid problems that could be related with recently introduced bugs (v2.1.0).
closed 2020-01-23 by jofemodo