zynthian/zynthian-sys #134: ON software update wifi is reset to not present

wyleu, 2019-12-27:
**Describe the bug**
ON software update wifi is reset to not present

**To Reproduce**
1. Select Wifi ON
2. Confirm Wifi functionality
3. Update Software
4. Confirm lack of Wifi functionality

**Expected behavior**
Wifi settings should be duplicated across updates

**Setup:**
HARDWARE
Raspberry Pi 3 Model B Rev 1.2
Soundcard: HifiBerry DAC+
Display: Generic HDMI Display
Wiring: MCP23017_EXTRA
GPIO Expander: MCP23017
SYSTEM
OS: Raspbian GNU/Linux 10 (buster)
Build Date: ???
Memory: 29% (269M/926M)
SD Card: 38% (11G/29G)
Temperature: 55.3ºC
MIDI
Profile: default.sh
Fine Tuning: 440 Hz
Master Channel: 16
QmidiNet: True
TouchOSC: False
SOFTWARE
zyncoder: master (8b7bfbd)
zynthian-ui: master (2aab6de)
zynthian-sys: master (3b70208)
zynthian-data: master (3d44cc8)
zynthian-webconf: master (74ca4d1)
LIBRARY
Snapshots: 2
User Presets: 9
User Soundfonts: 0
Audio Captures: 7
MIDI Captures: 0
NETWORK
Hostname: zynthian-amp2
Wifi: off
IP: 192.168.1.10 fdb4:3052:68a0:e300:2698:6c98:dc3f:2d93

Context:

very irritating on headless zynths as you don't have the GUI to re-enable wifi and you can't logon cos.... Your only option is reconnect using wired Ethernet.

jofemodo, 2019-12-28:
I just tried with a fresh BUSTER 2019-12-19, the last green build, and Wifi worked perfectly.
I'm testing with a v3 standard (RBPi3).

jofemodo, 2019-12-29:
It should be solved now. Please update & test!

commit 4e68927b (2019-12-29): Fix #134 & #121: WIFI problems on startup or when updating software / saving config.
closed 2019-12-29 by jofemodo