zynthian/zynthian-sys #139: XPolyMonk.lv2

Baggypants, 2020-01-22:
**Is your feature request related to a problem? Please describe.**
Add XPolyMonk.lv2 to list of synth engines.

**Describe the solution you'd like**
Include this install script as `/zynthian/zynthian-sys/scripts/recipes/install_XPolyMonk.sh`
```
#!/bin/bash

cd $ZYNTHIAN_PLUGINS_SRC_DIR
if [ -d "XPolyMonk.lv2" ]; then
        rm -rf "XPolyMonk.lv2"
fi
apt install -y libcairo2-dev libx11-dev
git clone --recursive https://github.com/brummer10/XPolyMonk.lv2.git
cd XPolyMonk.lv2
make
make install INSTALL_DIR=$ZYNTHIAN_PLUGINS_DIR/lv2
make clean
cd ..
rm -rf "XPolyMonk.lv2"
```

Baggypants, 2020-01-22:
https://github.com/zynthian/zynthian-sys/pull/140

jofemodo, 2020-01-22:
Curiously, it's classified as a "Generator", when it should be a Synthesizer ....

jofemodo, 2020-01-22:
CPU consumption is huge ....

Baggypants, 2020-01-22:
Yes it is. There are some notes about that here: https://linuxmusicians.com/viewtopic.php?f=24&t=20920#p113885

Baggypants, 2020-01-22:
That's also why it's categorised as 'generator'. It generates heat.

jofemodo, 2020-01-23:
;-DDD

commit 2fe27e92 (2020-01-22): Add exec permissions. Fixes #139.