zynthian/zynthian-sys #141: Forbidden 403 error on zynthian website is breaking build

guysoft, 2020-01-23:
```
--2020-01-22 22:59:07--  https://zynthian.org/download/pianoteq_stage_linux_trial_v660.7z
Connecting to zynthian.org (zynthian.org)|116.202.123.71|:443... connected.
HTTP request sent, awaiting response... 403 Forbidden
2020-01-22 22:59:07 ERROR 403: Forbidden.

7-Zip [32] 16.02 : Copyright (c) 1999-2016 Igor Pavlov : 2016-05-21
p7zip Version 16.02 (locale=C,Utf16=off,HugeFiles=on,32 bits,2 CPUs LE)

Scanning the drive for archives:

ERROR: No more files
/zynthian/zynthian-sw/pianoteq_stage_linux_trial_v660.7z

```

jofemodo, 2020-01-23:
But the file is available. Try:

`wget  https://zynthian.org/download/pianoteq_stage_linux_trial_v660.7z`

Regards,

jofemodo, 2020-01-24:
It seems to be working now ... probably a temporary problem.

guysoft, 2020-02-02:
```
 mkdir setbfree
+ ln -s /usr/local/share/setBfree/cfg/default.cfg ./setbfree
+ cp -a /zynthian/zynthian-data/setbfree/cfg/zynthian_my.cfg ./setbfree/zynthian.cfg
+ /zynthian/zynthian-sys/scripts/recipes/install_pianoteq_demo.sh
--2020-02-01 23:03:47--  http://zynthian.org/download/pianoteq_stage_linux_trial_v660.7z
Resolving zynthian.org (zynthian.org)... 116.202.123.71
Connecting to zynthian.org (zynthian.org)|116.202.123.71|:80... failed: No route to host.

7-Zip [32] 16.02 : Copyright (c) 1999-2016 Igor Pavlov : 2016-05-21
p7zip Version 16.02 (locale=C,Utf16=off,HugeFiles=on,32 bits,2 CPUs LE)

Scanning the drive for archives:

ERROR: No more files
/zynthian/zynthian-sw/pianoteq_stage_linux_trial_v660.7z

System ERROR:
Unknown error -2147024872
/bin/mv: cannot stat 'Pianoteq*': No such file or directory
```
Its failing constantly on this now.

closed 2021-07-05 by riban-bw