zynthian/zynthian-sys #144: infamousPlugins breaks build, again

guysoft, 2020-02-28:
Unlike #70, another reason this time
```
make[1]: *** [CMakeFiles/Makefile2:199: src/casynth/CMakeFiles/casynth_ui.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
[  2%] Built target infamous-rule
[  7%] Built target casynth
[  8%] Building CXX object src/casynth/CMakeFiles/casynth_ui.dir/casynth_ui_main.cxx.o
/zynthian/zynthian-sw/plugins/infamousPlugins/src/casynth/casynth_ui_main.cxx:185:1: error: invalid conversion from 'void* (*)(const _LV2UI_Descriptor*, const char*, const char*, LV2UI_Write_Function, LV2UI_Controller, void**, const LV2_Feature* const*)' {aka 'void* (*)(const _LV2UI_Descriptor*, const char*, const char*, void (*)(void*, unsigned int, unsigned int, unsigned int, const void*), void*, void**, const LV2_Feature* const*)'} to 'void* (*)(const LV2UI_Descriptor*, const char*, const char*, LV2UI_Write_Function, LV2UI_Controller, void**, const LV2_Feature* const*)' {aka 'void* (*)(const LV2UI_Descriptor*, const char*, const char*, void (*)(void*, unsigned int, unsigned int, unsigned int, const void*), void*, void**, const LV2_Feature* const*)'} [-fpermissive]
 };
 ^
make[2]: *** [src/casynth/CMakeFiles/casynth_ui.dir/build.make:73: src/casynth/CMakeFiles/casynth_ui.dir/casynth_ui_main.cxx.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:199: src/casynth/CMakeFiles/casynth_ui.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
/bin/mv: cannot stat '/zynthian/zynthian-plugins/bin/infamous-rule': No such file or directory
```

jofemodo, 2020-02-28:
Yes. I just pushed fixes for this problem, that was affecting to Artyfx and Infamous. Adding a "-fpermissive" to compiler options solved the build problem. Let's see if there is no segfaults ...  ;-)

Also pushed fixes for problems related to ZynaddSubFX and Shiro install scripts.

I hope tomorrow we can see the green again ...

Regards

jofemodo, 2020-03-04:
It's fixed

closed 2020-03-04 by jofemodo