zynthian/zynthian-sys #149: Zynthian permission denied

benwadub, 2020-03-23:
hi, i m new to Zynthian, when I try to login via ssh I get permission denied when using raspberry as password, I can access via webconf but it doesn't ask for login I can t access anything to try Zynthian if someone can help me?

jofemodo, 2020-03-23:
user: root
password: raspberry

zynthianOS runs everything as root, so only root user is active.

benwadub, 2020-03-23:
it s what I tried but didn't worked, I did an update on webconf and now I can access on my screen! thanks to tried to help me!

jofemodo, 2020-03-23:
It must work. What SD image are you using?

benwadub, 2020-03-23:
I dowloaded this 2019-06-26-zynthianos-stretch-lite-1.0.0-RC3
I still can t acmes to terminal via ssh, don t understand!

jofemodo, 2020-03-23:
Strange ... it should work ... could you try one of the latest buster builds?

benwadub, 2020-03-23:
I thought it was the latest! but to be honest I just tried with an cdmi screen, nothing else, maybe it s because of it? I just wanted to try Zynthian to see if I could use it to make an expander without screen, do you think it s possible just to plug a midi controller and audio output? or do I need a screen absolutely? if a screen is needed what could be the cheapest one?

jofemodo, 2020-03-24:
Hi @benwadub !

Please, this kind of questions are better addressed on the forum:

http://discourse.zynthian.org

BTW, you should connect the zynthian to the network and access with SSH.

Regards

benwadub, 2020-03-24:
no problem! sorry!
I just tried with the last sd image and I get the same error with ssh with Mac terminal, "permission denied"

mheidt, 2020-03-24:
Please share screenshots in the forum Tuesday,

benwadub, 2020-03-24:
[image]

that s what I get when writing my password with ssh in Mac osx terminal; I launched an update with the webconf to see if that permit me to access the ui like yesterday with the older image
thanks for the interest in my problem and sorry for my approximative English I m French

mheidt, 2020-03-24:
As @jodemodo said...root not zynthian has to be the user Tuesday,

benwadub, 2020-03-24:
I don t have to connect to Zynthian before login in root?

benwadub, 2020-03-24:
this is working thanks a lot!!! Zynthian seems to be a real beast!!!
do you know if with this corona story we can order on the Zynthian shop?

jofemodo, 2020-03-24:
Fantastic!
Regarding your question, you can order on the shop, but i'm not sure of the delivery time, as "Exception State" is declared on Spain. Probably you have to wait 1-2 weeks before we can send your kit.

benwadub, 2020-03-25:
Thanks, it ll let me time to fix my head hack with my midi din that don’t want to work :-)

closed 2020-03-31 by jofemodo