zynthian/zynthian-sys #158: Delay between start transport and CONTINUE message

riban-bw, 2020-06-02:
**Describe the bug**
There is a two second delay between starting the MIDI clock / JACK transport and a MIDI CONTINUE message being received. This makes step sequencer less responsive than expected.

**To Reproduce**
Steps to reproduce the behaviour:
1. Open Step Sequencer
2. Click SNAPSHOT encoder (or other method) to start playback
3. Click SNAPSHOT encoder (or other method) to pause playback
4. Click SNAPSHOT encoder (or other method) to resume playback

**Expected behaviour**
Playback starts immediately.

**Actual behaviour**
Playback starts after 2s when resuming.

**Setup:**
HARDWARE
Raspberry Pi 4 Model B Rev 1.2
Soundcard: HifiBerry DAC+ light
Display: WaveShare 7 HDMI+USB
Wiring: I2C_HWC
GPIO Expander: Not detected

YSTEM
Raspbian GNU/Linux 10 (buster)
Build Date: 2020-04-09
Memory: 6% (255M/3906M)
SD Card: 38% (9.8G/28G)
Temperature: 50.0ºC

SOFTWARE
zyncoder: master (319c9ab)
zynthian-ui: feature/stepseq (4781aa4)
zynthian-sys: master (790b139)
zynthian-data: master (9cccef5)
zynthian-webconf: master (3fee256)

**Additional context**
There is an optional parameter that may be passed to jack_midi_clock to specify this delay. By default the delay is 2s. `-d0` will make this delay 0s.

jofemodo, 2020-06-10:
Hi @riban-bw !

Could we close this?

riban-bw, 2020-06-11:
The reason I didn't close this immediately is because I made the change manually to my `/etc/systemd/system/jack-midi-clock.service` and wasn't sure how this is updated by zynthian-sys. I see this happens during a system update.

I reset my file, switched to master branch, performed update via webconf and it updated correctly. I then switched back to feature/stepseq branch and MIDI clock was responding as expected. Thanks!

closed 2020-06-11 by riban-bw