zynthian/zynthian-sys #165: Zynthian fork of mod-ui is very old

steveb, 2020-07-05:
https://github.com/zynthian/mod-ui is lagging upstream https://github.com/moddevices/mod-ui by about 2 years now. It would be useful to have a branch created which tracks upstream master so we can start applying our customization commits.

Maybe a more conservative approach is needed on the amount and complexity of customizations so that the burden of keeping up with upstream master is lower.

guysoft, 2020-07-06:
Kind of like #55 which is from two years ago.
It seem like it makes sense to auto-update it rather than needing to open an issue each time for it to update.

steveb, 2020-07-06:
Actually the series of changes apply fairly cleanly against https://github.com/moddevices/mod-ui, I've posted the branch here https://github.com/steveb/mod-ui/tree/zyn-mod-merge

I haven't tried running it yet

closed 2020-07-07 by steveb