zynthian/zynthian-sys #169: Migrate to 64 bits

jofemodo, 2020-07-16:
I've created this thread to discuss and coordinate the technical aspects of migrating ZynthianOS to 64 bits.
Probably, it won't be a straightforward task, so let's start by drafting a route sheet and proceed step by step.

I would assume that we will keep using "Raspbian", but i'm not closed to use other distros if we have good reason to do so.

1. Test Raspbian 64 bits and the key pieces: jackd, jalv and standalone engines installed from repository
2. Test hardware drivers: hifiberry, piscreen, audioinjector, USB compliant devices, etc.
3. Test Zynthian UI with standalone engines and some LV2 plugins
4. Modify and test the "SD setup scripts". Start building Buster64 SD images.
5. Test, test and test ...

Regards!

jofemodo, 2020-07-16:
@guysoft , how is working the CustomPiOS system with 64 bits images? Could you give some advice regarding the task?

riban-bw, 2020-07-16:
Remember that pianoteq only provide 32-bit ARM build so the 32-bit libs are required as detailed in the README.

guysoft, 2020-07-16:
Hey, a bit of clarification:
1. There is no raspbian arm64/aarch, only raspberry Pi OS, see: https://www.tomshardware.com/news/raspberry-pi-os-no-longer-raspbian
2. There is no lite version of RaspberryPiOS, see: 
https://community.octoprint.org/t/is-it-really-true-that-octoprint-does-not-support-4gb-version-of-pi4/21852/13?u=guysoft 
3. Might be a reason ro consider Ubuntu arm64bit or wait more

guysoft, 2020-11-04:
Update: there is now a 64bit lite version, latest CustomPiOS has a new variant ``raspios_lite_arm64``

https://github.com/guysoft/CustomPiOS/releases/tag/1.4.0

teknico, 2021-01-12:
> Remember that pianoteq only provide 32-bit ARM build so the 32-bit libs are required as detailed in the README.

Pianoteq 7.1.0 now [includes](https://www.modartt.com/pianoteq#changes) 64-bit Linux ARM binaries.

guysoft, 2021-01-12:
It seems like Raspberrypi foundation are not maintaining the 64 bit build. See: https://github.com/RPi-Distro/pi-gen/issues/481

jofemodo, 2021-01-12:
OK @guysoft ! Nice to know. We have not a serious plan for migrating yet, so this delay is not a big problem for us. Anyway, we have to migrate sooner than later, so please, keep us informed ...

Thanks a lot!

mkesper, 2022-03-18:
Why not build upon Debian Bullseye itself? That's what Raspberry Pi OS uses as base for 64bit: https://downloads.raspberrypi.org/raspios_arm64/release_notes.txt

guysoft, 2022-03-20:
@mkesper I am not sure I understand your suggestion. What do you mean by "upon Debian Bullseye itself"? The current builds upon Raspberry Pi OS.  When this issue was opened there was no Raspberry Pi OS 64bit. Now there is one but its unstable.

mkesper, 2022-03-27:
Sorry, was confused. Right, Raspberry Pi OS includes Debian and everything the Pi needs to boot (firmware). Maybe it could be made a config option to do a "pioneer" build based on the 64bit Pi OS?

guysoft, 2022-03-27:
@mkesper I think what you are describing is already possible in CustomPiOS, you can build a variation on an existing build, known as a ``variant``. In many cases its used to build the OS on a different architecture. However, that is not the limitation in order to have ZynthianOS working on 64bits. The real limitation is 
1.  Raspberry Pi OS is still beta and many things don't work, see https://github.com/RPi-Distro/pi-gen/issues/481#issuecomment-745153854 its from 2020, its still the case. We have, for example, decided to use the Ubuntu image for the 64bit build of OctoPi.
2. It requires things on zynthian's part to be built and tested for 64bit, its a lot of packages and it something that would require considerable work by @jofemodo and other.

Kirtai, 2022-08-14:
The 64bit version is now out of beta.

tunagenes, 2023-05-02:
I have used 64 bit RaspianOS with Jack, fluidsynth. and PureData (left out the space in Pure Data intentionally).  Let me know if there is anything specific you'd like tested.  I think PureData is (compiled for} a 32bit program, and fluidsynth 64bit, as I am running them.

riban-bw, 2023-05-05:
Yes, I too am running  various audio modules on 64-bit Bullseye (aarch64) without issue but Zynthian has many modules that all need to be ported and tested. I suggest we configure a 64-bit nightly build in CI so that we can get reports of what fails to build. We can then chip away at the failures until we get an image built and give the community the opportunity to start testing. I am unfamiliar with the build mechanism and CI for Zynthian and have no spare capacity to learn right now but maybe one of our image-build experts might offer some guidance.

riban-bw, 2024-03-12:
64-bit image is now default for oram, the current testing version.

closed 2024-03-12 by riban-bw