zynthian/zynthian-sys #170: Zynthian webconf forces cmdline.txt to use root=/dev/mmcblk0p2

Baggypants, 2020-07-20:
**Describe the bug**

Testing boot from usb on pi4, I got it working until I chose the kit in the zynthian webconf then it failed to boot. I tracked it down the cmdline.txt `root=` directive changed from `root=/dev/sda2` back to `root=/dev/mmcblk0p2`

**To Reproduce**
Steps to reproduce the behavior:

1. Change cmdline.txt root= directive to anything with a text editor
2. Pick a kit and save
3. review root= directive

**Expected behavior**
Somehow you're going to have to detect what device is mounted as root and extract that device `mount | grep ' / ' | awk '{print $1}` might be a place to start!

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Setup:**
 - SD-image
 - Hardware Config
 - Software Version

**Additional context**
Add any other context about the problem here.

jofemodo, 2020-11-21:
Hi @Baggypants !

Read this:

https://discourse.zynthian.org/t/disable-update-of-zynthian-service-and-creation-of-99-fbdev-conv/4999

I think it gives a solution for this issue. 

Regards,

jofemodo, 2020-11-27:
Hi @Baggypants ! Do you think we can close the issue? Or do you want to test it on your use-case?

closed 2021-08-12 by jofemodo