zynthian/zynthian-sys #172: Audio Injector Config needs Webconf config edit to 'activate' card.

wyleu, 2020-07-30:
**Describe the bug**
Audio Injector needs Webconf config edit to 'activate' card. When installing an Audio Injector card installation the card will not activate until a small edit ( addition of a space into the admin sections seems to work)

**To Reproduce**
Install Audio Injector Card
1. Install zynthian software.
2. Select audio hardware (Custom/Audio select Audio Injector Card)
3. Reboot machine
4. Login in to zynthian 
5. Run alsa Mixer 
6. Tab To Capture 
7. Select Capture L - R 
8. close alsamixer
9. Reboot.
10. No Audio output ( Audio Test) 
11. Select audio hardware (Custom/Audio select Audio Injector Card)
12. Edit custom settings 
13. Add space
14. Save
15. Reboot
16 Audio Output confirmed (Audio Test)

**Expected behavior**
An Audio INjector Card should play Audio Test upon installation

**Screenshots**
Not at the moment :-)

**Setup:**
- Not available.

**Additional context**
Been a little dance required for audio injector for a while now.
This processs does NOT confirm Audio input.

jofemodo, 2020-07-30:
Hi @mheidt !
I just fixed the onboot config script for building the zyncoder. Please, test again ...

closed 2020-07-30 by jofemodo

jofemodo, 2020-08-07:
Hi @mheidt !
Please, don't forget to test this when you have some time ...

commit f61575eb (2020-07-30): Fix #172: Build zyncoder when updating envars on boot.