zynthian/zynthian-sys #174: Use alsa_out to enable "headphone" HiFiberry output

Baggypants, 2020-08-07:
**Is your feature request related to a problem? Please describe.**
The new zyn4 kit shows "Headphone" on the onboard rpi output. It does nothing by default

**Describe the solution you'd like**
- re-add `dtparam=audio=on` to `/boot/config.txt` to expose the onboard audio
- use `alsa_out -j Headphones -d hw:Headphones` to create a jack port to send audio to the onboard hardware jack.
- edit zynthian_autoconnect to automatically add outputs from audio chains to Headphones as well as jackpeak and the system out

This has been tested working on a pi4 using a HiFi Berry. The audio is sent to both outputs simultaneously.

**Describe alternatives you've considered**
purchasing a headphone amp

jofemodo, 2020-08-07:
OK! It's implemented and it works like a charm. You can enable/disable the headphones output from the UI's admin menu. I preferred to add the option because alsa_out takes 5% of 1 CPU Core and some people would prefer to disable when not used.

Thanks a lot for the tips, @Baggypants !

riban-bw, 2020-08-07:
Not working for me. Output from journalctl:

```
Aug 07 14:15:32 zynthian1 systemd[1]: headphones.service: Service RestartSec=1s expired, scheduling restart.
Aug 07 14:15:32 zynthian1 systemd[1]: headphones.service: Scheduled restart job, restart counter is at 4.
Aug 07 14:15:32 zynthian1 systemd[1]: Stopped Extra jack output for RBPi Headphones.
Aug 07 14:15:32 zynthian1 systemd[1]: Starting Extra jack output for RBPi Headphones...
Aug 07 14:15:32 zynthian1 jack_wait[716]: server is available
Aug 07 14:15:32 zynthian1 systemd[1]: Started Extra jack output for RBPi Headphones.
Aug 07 14:15:32 zynthian1 alsa_out[719]: ALSA lib pcm_hw.c:1822:(_snd_pcm_hw_open) Invalid value for card
Aug 07 14:15:32 zynthian1 alsa_out[719]: Capture open error: No such device
Aug 07 14:15:32 zynthian1 systemd[1]: headphones.service: Main process exited, code=exited, status=20/n/a
Aug 07 14:15:32 zynthian1 systemd[1]: headphones.service: Failed with result 'exit-code'.
Aug 07 14:15:33 zynthian1 systemd[1]: headphones.service: Service RestartSec=1s expired, scheduling restart.
Aug 07 14:15:33 zynthian1 systemd[1]: headphones.service: Scheduled restart job, restart counter is at 5.
Aug 07 14:15:33 zynthian1 systemd[1]: Stopped Extra jack output for RBPi Headphones.
Aug 07 14:15:33 zynthian1 systemd[1]: headphones.service: Start request repeated too quickly.
Aug 07 14:15:33 zynthian1 systemd[1]: headphones.service: Failed with result 'exit-code'.
Aug 07 14:15:33 zynthian1 systemd[1]: Failed to start Extra jack output for RBPi Headphones.
```

jofemodo, 2020-08-07:
Hi @riban-bw ,

What SD image are you using?
Please, send me the output of "aplay -l".

Regards,

riban-bw, 2020-08-07:
This image is running image with build date 2020-04-09. aplay -l:
```
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 IEC958/HDMI [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: ALSA [bcm2835 ALSA], device 2: bcm2835 IEC958/HDMI1 [bcm2835 IEC958/HDMI1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: UMC1820 [UMC1820], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
```

My other image is running build with date 2020-05-27. aplay -l:
```
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 IEC958/HDMI [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: ALSA [bcm2835 ALSA], device 2: bcm2835 IEC958/HDMI1 [bcm2835 IEC958/HDMI1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: UMC1820 [UMC1820], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
```

Since updating the later build (RC2) I can't access Soundcard in webcong (500: Internal Server Error) so I can't get into it to check how well it works. All a bit frustrating!!!

jofemodo, 2020-08-08:
It should be fixed now. Please, update, reboot and test again ...

riban-bw, 2020-08-08:
Sorry,, not yet. Now there is no headphone option in admin menu.

jofemodo, 2020-08-08:
Try now ...

closed 2020-08-09 by jofemodo

riban-bw, 2020-08-11:
alsa_out is consuming 6% CPU on RPi4 which feels rather excessive. Adding `-q0` to alsa_out command reduces CPU usage to 3% (which isn't ideal but twice as good as default setting) and quality reduction is imperceptible to me listening to Pianoteq blues demo. (Setting -q1 did cause aliasing which is odd.) I am not sure if there is other optimisation possible but I would recommend adding `-q0` to the service script.

jofemodo, 2020-08-11:
Done! 7eb1e14