zynthian/zynthian-sys #175: epdf_detect.sh error

jofemodo, 2020-08-08:
From the forum:
```

While we’re testing:

When I’m running update (rc2, v4 kit) I’m getting the following error:
/zynthian/zynthian-sys/scripts/epdf_detect.sh: line 7: [: too many arguments

This script needs some quotes as the spaces in the following file are causing problems:
cat /proc/device-tree/hat/product
DAC+ ADC Pro
```

https://discourse.zynthian.org/t/the-zynthian-kit-v4-has-been-released/4419/85

jofemodo, 2020-08-08:
This is for you @smiths73v3. I sent you an invitation.

smiths73v3, 2020-08-09:
I'll take a look. First glance at the report looks like the DAC+ ADC Pro has an eeprom that is getting reported in the device tree.

smiths73v3, 2020-08-09:
Fix committed.
Forgot to mention #175 in the commit message:

https://github.com/zynthian/zynthian-sys/commit/b9e11a9179f211f31c3d686f334f4b98a75d2308#diff-d6c5855a62cf32a4dadbc2831f0f295f

closed 2020-08-09 by smiths73v3