zynthian/zynthian-sys #185: Nightly Builds - Persistent xruns

gr0k-net, 2021-02-14:
**Describe the bug**
Although the stable version of Zynthian OS works fine on my Zynthian after moving to the feature/stepseq branch, I have not been able to get any of the nightly builds working. I have tried several, with the latest being from 2021-01-28. 

The error I get in the UI Log is hundreds of lines like this:

Feb 02 03:48:55 zynthian startx[505]: WARNING:zynthian_autoconnect.cb_jack_xrun: Jack Audio XRUN! 

This begins immediately after booting, even tho no layers are loaded. Adding layers doesn't fix this issue.

Additionally, nightly builds don't appear to recognise my MPK Mini Mk2 USB midi controller, completely ignoring input regardless of channel. I have tried both sets of USB ports, and although power is provided to the keyboard, no notes are played, and no 'm' icon appears in the status area.

Whilst troubleshooting these issues, I've tried many things, including limiting USB speed to 12Mb/s, and the config.txt edit suggested here https://github.com/zynthian/zynthian-issue-tracking/issues/302 (although in my instance, I added the line total_mem=8192) Nothing has worked so far, the xruns continue, and my USB keyboard is not recognised.

The output of aconnect -i :
client 0: 'System' [type=kernel]
    0 'Timer           '
    1 'Announce        '
client 14: 'Midi Through' [type=kernel]
    0 'Midi Through Port-0'
client 16: 'f_midi' [type=kernel,card=0]
    0 'f_midi          '

In webconf / hardware / wiring, s1-4 default to midi prog change 1-4. Pressing these buttons shows the m midi icon in the status area as expected, but Zynthian continues to ignore the keyboard.

This a brand new Zynthian 4.1 kit with an 8gb Pi 4. I have tried with a several SD cards, including a brand new never been used 64GB SanDisk Extreme. I am using the official Raspberry Pi power kit. This has been plugged in directly, and via UPS with the same results. Powering via battery pack made no difference either.

I can use the device as expected on the stable image.

**To Reproduce**
Steps to reproduce the behavior:
1. Download and image latest nightly build (most recently tested with 2021-01-28, tho have tried 2021-01-26, and several others) to SD card
2. Insert into Zynthian and boot
3. Experience xruns immediately after finished booting
4. Update Zynthian using web ui, and restart
5. Continue to experience xruns
6. Swap to feature/stepseq branch, update, and restart
7. Continue to experience xruns
8. Add instruments and use device
9. Continue to experience xruns

**Expected behavior**
Being able to use the latest image on the device without xruns.

**Setup:**
 - SD-image: 2021-01-28 (tho have tried others)
 
 - Hardware Config: Official v4.1 Kit, 8GB Pi 4, Official Pi power pack, SanDisk Extreme SD card (tho have tried others)
Raspberry Pi 4 Model B Rev 1.4
Soundcard: HifiBerry DAC+ ADC PRO
Display: ZynScreen 3.5 (v1)
Wiring: MCP23017_ZynScreen
GPIO Expander: MCP23017

 - Software Version:
zyncoder: master (3bb3969)
zynthian-ui: feature/stepseq (ce0c90f)
zynthian-sys: master (d8acbf3)
zynthian-data: master (336eadc)
zynthian-webconf: master (0587018)

riban-bw, 2021-03-31:
I had a fairly stable system with xruns very infrequently but something has changed and it now xruns with even simple engines loaded. We need to investigate why. I feel this is the highest priority issue. We should not have a stable system which clicks.

jofemodo, 2021-03-31:
Hi @riban-bw !

Could you give more details? Have you been changing from stable, testing, development branches, etc.? Or it was a totally "stable" system? Did you tried to update or change anything on the config?

Thanks

riban-bw, 2021-04-01:
I don't remember the exact sequence of events. This is my V4.2 which I try to keep on stable but have switched to testing a few times. I have also switched individual repos to dev branches. It seemed to happen after I experimented with various plugins or system to use audio input. Frankly I don't have confidence in any of these memories so probably can't attribute the onset of the issue to a particular activity.

jofemodo, 2021-04-04:
Currently, when you change to testing, you can't return to the stable 100%. The update recipes from zynthian-sys are one-way only and can't be reverted (currently).

This should be warned when leaving stable for first time.

riban-bw, 2021-07-05:
Is this still an issue? Recent fixes for libzynseq may have (partially) resolved this.

jofemodo, 2021-07-05:
Nightly builds are currently stopped as we are using the update script mechanism for releasing the stable SD versions.
We can close this.

closed 2021-07-05 by jofemodo