zynthian/zynthian-sys #186: Critical: Software update modifies / corrupts .vnc/passwd

mark-orion, 2021-02-23:
**Describe the bug**
Whenever the software update is run, any existing VNC configuration becomes unusable because the password file .vnc/passwd is modified and any existing authorization credentials are lost. I consider this critical as a remotely controlled system becomes unusable. 

**To Reproduce**
Steps to reproduce the behavior:
login via ssh and check the modification time of .vnc/passwd
run software update
the file has changed

**Expected behavior**
any existing .vnc/passwd should not be touched.

jofemodo, 2021-02-23:
It should be fixed now. Please, update and test.

closed 2021-02-23 by jofemodo

mark-orion, 2021-02-24:
I can confirm that the problem is gone. Thanks for fixing this so quickly.

commit b953240c (2021-02-23): Fix #186: Typo was resetting vnc password on every update.