zynthian/zynthian-sys #192: Panel to manage windows in VNC display

riban-bw, 2021-03-25:
**Is your feature request related to a problem? Please describe.**
In the VNC extended display the windows are managed by xfce window manager. This provides window decoration and other window management elements. It may be advantageous to be able to launch applications and see what is running.

**Describe the solution you'd like**
A panel (task bar) displayed in VNC viewer which allows launching of applications, e.g. patchage and control and monitoring of running applications.

**Describe alternatives you've considered**
A panel or dock which shows the avialable applications and allows starting and focusing them.

**Additional context**
I have worked up an example using xfce4-panel which has a panel at the top of the screen. The panel has a combined desktop switcher / appliation selector drop down, launch buttons for each of the required applications (patchage and terminal), indications of running applications and a clock. I wrote a helper script which only launches an application is it is not already running so that the launcher buttons can be used without triggering lots of instances of applications.

[image]

riban-bw, 2021-03-25:
Oops! I'm not completely happy with this implementation. I was considering changing the panel to be smaller hence I wanted to do more testing hence delay before PR. Maybe we should be more stringent on the workflow, requiring a PR before any merges take place to avoid partial developments being merged too early.

jofemodo, 2021-03-25:
You are right. No more merges without a PR ;-)
For this case, i would like to see improved the "look and feel" of launcher buttons. They are really ugly. In fact we should try to keep a coherent style on all zynthian interfaces.

Thanks!

riban-bw, 2021-03-25:
One of these days I might take offence at you describing my work as "ugly" :wink: .

I looked at various docks but couldn't get any to work. I observe the choice of xfce4 as the window manager so used the xfce4 panel. I could not find a combined launcher and running app manager plugin so effectively wrote one by calling a helper script to launch apps. In practice we only need to launch a couple:

- Patchage
- Terminal

Other running apps are engines launched by Zynthian. I added the desktop / app widget which:

- Is small
- Indicates the currently focused app
- Allows focusing of an app
- Allows moving apps between desktops
- Some other features

I quite like this because it means we could lose the currently running app widget and collapse down to a smaller panel. We could use icons for the launchers and recover most of the top line of the screen.

jofemodo, 2021-03-30:
Ohhh! Please, don't do that! Your work is beautiful, but the launcher buttons are ugly. It's not your fault, it's the buttons fault.
I'm pretty sure we can add some decoration to this buttons and convert the ugly into beauty.

BTW, I would remove these elements:

+ clock (a cpu/memory monitor will be more useful)
+ "window manager" buttton
+ "panel config" button

We only need the "patchage" and the "terminal" buttons. Less is more ;-)

Thanks!

riban-bw, 2021-03-30:
Yep! Those were on my list for the chop. I think just the pull-down window manager so that users can access hidden apps and Lainey's for Patchage and terminal. We could also use icons instead of words for the launcher buttons which may be less ugly. I will have a look.

riban-bw, 2024-03-12:
Things may not have gotten much less _ugly_ but it is working fine so closing this ticket.

commit 3701dc26 (2021-03-25): Fix #192: Install patchage.
closed 2024-03-12 by riban-bw