zynthian/zynthian-sys #202: Use package management for system updates

riban-bw, 2021-05-19:
**Is your feature request related to a problem? Please describe.**
ZynthianOS currently uses a mixture of scripts to update the system which access github repositories directly. This has an overhead of requiring recompilation of some modules on the target machine. Any binaries required for such updates are not best placed in a git repository. Running apt upgrade on a Zynthian will break the OS. A common issue is users running apt upgrade and breaking their Zynthians. It can be a public relations embarasement to tell them they have done something that they feel was reasonable.

**Describe the solution you'd like**
Create and use a package management system to deploy updates to ZynthianOS. The obvious choice is to use .deb packages because ZynthianOS is based on Debian and Debian package management is established, well tested and effective. ZynthianOS should be configured to use the Zynthain deb repo by default and also be locked to certain package versions on either repo where there are incompatibilities with later versions.

**Describe alternatives you've considered**
Continue using git for updates and tell users to not apt upgrade. Accept updates may take longer than expected and have a risk of failing with difficult support issues.

**Additional context**
This is already agreed in principle. This ticket is to formalise the request and give context to it to better understand some of the benefits. We can track progress of this effort here.

jofemodo, 2021-05-20:
Hi @riban!

Test the deb-zynthian branch for zynthian-sys. When updating, the new zynthian repo should be installed and the alsa-utils package, updated.

Regards,

riban-bw, 2021-05-20:
I am not seeing this. I switched to deb-zynthian branch and ran zynthian update `apt edit-sources` shows:

```
deb http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi
deb http://www.deb-multimedia.org buster main non-free
```

`apt-update` gives:

```
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:2 http://archive.raspberrypi.org/debian buster InRelease [32.9 kB]
Hit:3 http://www.deb-multimedia.org buster InRelease
Hit:4 http://ppa.launchpad.net/kxstudio-debian/libs/ubuntu bionic InRelease
Hit:5 http://ppa.launchpad.net/kxstudio-debian/music/ubuntu bionic InRelease
Hit:6 http://ppa.launchpad.net/kxstudio-debian/plugins/ubuntu bionic InRelease
Hit:7 http://ppa.launchpad.net/kxstudio-debian/apps/ubuntu bionic InRelease
Hit:8 http://ppa.launchpad.net/kxstudio-debian/kxstudio/ubuntu bionic InRelease
Ign:9 https://kx.studio/repo stable InRelease
Ign:10 https://kx.studio/repo gcc5 InRelease
Hit:11 https://kx.studio/repo stable Release
Hit:12 https://kx.studio/repo gcc5 Release
Get:13 http://archive.raspberrypi.org/debian buster/main armhf Packages [376 kB]
Fetched 423 kB in 3s (131 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
205 packages can be upgraded. Run 'apt list --upgradable' to see them.
```

riban-bw, 2021-05-20:
I see the new zynthian repo (how exciting) and it has installed:

- alsa-utils
- libasound2
- libasound2-data
- libasound2-dev
- libasound2-doc
- libasound2-plugin-smixer
- libatopology-dev
- libatopology2

But `apt update` still shows 224 packages that can be upgraded (which may break Zynthian):
```
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Hit:3 http://www.deb-multimedia.org buster InRelease
Hit:4 http://os.zynthian.org:8090/zynthian-testing buster InRelease
Reading package lists... Done
Building dependency tree
Reading state information... Done
224 packages can be upgraded. Run 'apt list --upgradable' to see them.
```

jofemodo, 2021-05-20:
I just tested with a fresh burned image. Simply changed to testing, and selected the deb-zynthian on zynthian-sys repo. Then run "update_zynthian.sh". It worked smoothly and installed the new alsa-utils. Have you checked if alsa-utils are updated to 1.2.4?

Regarding the 224 packages to upgrade, it's totally normal, but you don't want to update the system, only the needed packages. Anyway, it also did a full upgrade (after), and the zynthian is working perfectly. The problem seems to be solved ...

Regards,

jofemodo, 2021-05-21:
Hi @riban!

I just rebuild the alsa-utils deb package. It was not correctly generated. 

Regards,

riban-bw, 2021-05-21:
We probably don't actually need alsa-utils 1.2.4 but I guess no harm in keeping these packages fairly up-to-date. (alsa-lib is the one that is broken before 1.2.4.)

Regarding 224 packages, if I read you correctly, you say that the Zynthian update mechanism does not currently perform a full upgrade but that a manual run of `apt upgrade` does not break Zynthian (like it used to do). It would be good to get to the stage where we are confident that an apt update can be run without breaking Zynthian and include it in the update mechanism but this might require more careful control of packages, e.g. if upstream change a package which breaks Zynthian then a user performs an update before we have a chance to catch it we could add to the support burden. Maybe CI may help with this.

jofemodo, 2021-05-21:
Step by step. I hope to merge this on testing in the next days. The backend part (deb repository) needs a little bit more of work before going production.

jofemodo, 2021-05-27:
It's merged on testing.
The deb-repo is fully working now! ;-)

Awaiting for more packages, we need to define the workflow for managing packages.

Regards,

jofemodo, 2021-07-01:
The deb repo infrastructure is running OK. Next steps:

+ Start creating deb packages for every install recipes. In fact we should generate a new list of "deb-build recipes". Funny task! :-)
+ Implement a cross-compiling "deb builder" task and add it to Jenkins tasks.
+ Modify SD setup script for using deb packages instead of install recipes.

riban-bw, 2021-07-06:
I will close the issue about the risk of uncontrolled third party repos but we must ensure we track them here, i.e. move all third party dependencies t deb packages. This may require local clones from upstream or a script that creates the deb which is run on-demand (not automatically by update mechanism) so that release management can validate any changes.

riban-bw, 2024-03-12:
Closing this issue as deb repo is implemented and we are not actually tracking stuff here.

closed 2024-03-12 by riban-bw